dingjikerbo/Android-BluetoothKit #4: library里面AndroidManifest的supportsRtl属性建议去掉

taoyuanxiaoqi, 2016-11-17:
不支持阿语语系右对齐的应用 引用库的时候会导致编译不过 求去掉改标志

> Error:注: 某些输入文件使用或覆盖了已过时的 API。
> 注: 有关详细信息, 请使用 -Xlint:deprecation 重新编译。
> 注: 某些输入文件使用了未经检查或不安全的操作。
> 注: 有关详细信息, 请使用 -Xlint:unchecked 重新编译。
> 
> 注: /Users/lee/source/taoxiaoqiPublic_android/sociallogin/src/main/java/com/taoxiaoqi/activity/T4JTwitterLoginActivity.java使用或覆盖了已过时的 API。
> 注: 有关详细信息, 请使用 -Xlint:deprecation 重新编译。
> /Users/lee/source/taoxiaoqiPublic_android/taoxiaoqi/src/main/AndroidManifest.xml:39:9-36 Error:
> 	Attribute application@allowBackup value=(false) from AndroidManifest.xml:39:9-36
> 	is also present at [com.inuker.bluetooth:library:1.2.3] AndroidManifest.xml:21:9-35 value=(true).
> 	Suggestion: add 'tools:replace="android:allowBackup"' to <application> element at AndroidManifest.xml:37:5-269:19 to override.
> /Users/lee/source/taoxiaoqiPublic_android/taoxiaoqi/src/main/AndroidManifest.xml:44:9-36 Error:
> 	Attribute application@supportsRtl value=(false) from AndroidManifest.xml:44:9-36
> 	is also present at [com.inuker.bluetooth:library:1.2.3] AndroidManifest.xml:23:9-35 value=(true).
> 	Suggestion: add 'tools:replace="android:supportsRtl"' to <application> element at AndroidManifest.xml:37:5-269:19 to override.
> FAILURE: Build failed with an exception.
> 
> * What went wrong:
> Execution failed for task ':taoxiaoqi:processDailyDebugManifest'.
> > Manifest merger failed with multiple errors, see logs
> 
> * Try:
> Run with --stacktrace option to get the stack trace. Run with --debug option to get more log output.
>

taoyuanxiaoqi, 2016-11-17:
还有这个属性 

`android:allowBackup="true"`

dingjikerbo, 2016-11-18:
好的，这个可以去掉

dingjikerbo, 2016-11-21:
你更新到1.2.4吧

在 2016年11月18日 下午2:23，Wentian Li <dingjikerbo@gmail.com>写道：

> 好的，这个可以去掉
>

taoyuanxiaoqi, 2016-11-21:
好的 谢谢

closed 2016-11-21 by taoyuanxiaoqi