dingjikerbo/Android-BluetoothKit #8: 初始化出现nullpointer exception

taoyuanxiaoqi, 2016-11-28:
> Could not dispatch event: class com.taoxiaoqi.android.device.event.DeviceListChangeEventModel to subscribing class class com.taoxiaoqismart.stencil.presenter.home.DeviceListFragmentPresenter
>                                                      java.lang.NullPointerException: Attempt to invoke virtual method 'android.content.Intent android.content.Context.registerReceiver(android.content.BroadcastReceiver, android.content.IntentFilter)' on a null object reference
>                                                          at com.inuker.bluetooth.library.utils.BluetoothUtils.registerGlobalReceiver(BluetoothUtils.java:51)
>                                                          at com.inuker.bluetooth.library.utils.BluetoothUtils.registerReceiver(BluetoothUtils.java:47)
>                                                          at com.inuker.bluetooth.library.receiver.BluetoothReceiver.<init>(BluetoothReceiver.java:67)
>                                                          at com.inuker.bluetooth.library.receiver.BluetoothReceiver.getInstance(BluetoothReceiver.java:56)
>                                                          at com.inuker.bluetooth.library.BluetoothClientImpl.registerBluetoothStateReceiver(BluetoothClientImpl.java:528)
>                                                          at com.inuker.bluetooth.library.BluetoothClientImpl.registerBluetoothReceiver(BluetoothClientImpl.java:522)
>                                                          at com.inuker.bluetooth.library.BluetoothClientImpl.<init>(BluetoothClientImpl.java:118)
>                                                          at com.inuker.bluetooth.library.BluetoothClientImpl.getInstance(BluetoothClientImpl.java:127)
>                                                          at com.inuker.bluetooth.library.BluetoothClient.<init>(BluetoothClient.java:31)
>                                                          at com.taoxiaoqi.ble.BLEToolManager.<init>(BLEToolManager.java:35)

1.2.5版本

taoyuanxiaoqi, 2016-11-28:
老版本无此问题

dingjikerbo, 2016-11-28:
谢谢提出问题，这个尽快修复

dingjikerbo, 2016-11-28:
已修复，请更新到1.2.6

closed 2016-11-29 by dingjikerbo

dingjikerbo, 2016-12-07:
用最新版本吧1.2.9