dingjikerbo/Android-BluetoothKit #9: write接口速度问题

taoyuanxiaoqi, 2016-12-07:
> 12-07 11:45:19.044 15540-15540/com.taoxiaoqi D/bleSpeed: call write.. 00 00 00 
> 12-07 11:45:19.224 15540-15540/com.taoxiaoqi D/bleSpeed: write.. onResponse 0
> 12-07 11:45:19.374 15540-15540/com.taoxiaoqi D/bleSpeed: call write.. 00 30 10 4F 66 A5 59 02 B0 3E DB 18 DC 8C CA 42 C1 59 4E 37 
> 12-07 11:45:19.564 15540-15540/com.taoxiaoqi D/bleSpeed: write.. onResponse 0
> 12-07 11:45:19.564 15540-15540/com.taoxiaoqi D/bleSpeed: call write.. 01 37 65 06 3E 21 58 4F 06 4E EC 05 8B A8 82 D5 70 6D 8A A0 
> 12-07 11:45:19.764 15540-15540/com.taoxiaoqi D/bleSpeed: write.. onResponse 0
> 12-07 11:45:19.764 15540-15540/com.taoxiaoqi D/bleSpeed: call write.. 02 1D DD 4F 7A 92 75 B0 2D B4 7C 71 78 
> 12-07 11:45:19.954 15540-15540/com.taoxiaoqi D/bleSpeed: write.. onResponse 0

请教个问题，最近在测试性能的时候，发现写character的时候，从写入到接到response大概都要将近200ms的时间。没有其他ble的调试经验，IOS项目基本上60ms一个包就能搞定了，我们这个速度是不是有点慢？

dingjikerbo, 2016-12-07:
你用android原生接口试试

taoyuanxiaoqi, 2016-12-07:
> 12-07 14:06:31.444 15211-15211/com.example.android.bluetoothlegatt D/bleSpeed: bt_query send a data..
> 12-07 14:06:31.534 15211-15231/com.example.android.bluetoothlegatt D/bleSpeed: onCharacteristicWrite6
> 12-07 14:07:17.234 15211-15211/com.example.android.bluetoothlegatt D/bleSpeed: bt_query send a data..
> 12-07 14:07:17.314 15211-15229/com.example.android.bluetoothlegatt D/bleSpeed: onCharacteristicWrite6
> 12-07 14:07:24.564 15211-15211/com.example.android.bluetoothlegatt D/bleSpeed: bt_query send a data..
> 12-07 14:07:24.774 15211-15229/com.example.android.bluetoothlegatt D/bleSpeed: onCharacteristicWrite6
> 
> 12-07 14:26:42.434 20690-20690/com.inuker.bluetooth D/bleSpeed: call write ...
> 12-07 14:26:42.544 20690-20690/com.inuker.bluetooth D/bleSpeed: onResponse -1
> 12-07 14:27:18.784 20690-20690/? D/bleSpeed: call write ...
> 12-07 14:27:18.894 20690-20690/? D/bleSpeed: onResponse -1
> 12-07 14:27:22.704 20690-20690/? D/bleSpeed: call write ...
> 12-07 14:27:22.814 20690-20690/? D/bleSpeed: onResponse -1
> 12-07 14:27:34.424 20690-20690/? D/bleSpeed: call write ...
> 12-07 14:27:34.534 20690-20690/? D/bleSpeed: onResponse -1

我分别测试了google的官方demo和您的demo，发现速度差不多 都是80～90ms左右，我还是去检查一下我的使用方式吧。

closed 2016-12-07 by taoyuanxiaoqi