dingjikerbo/Android-BluetoothKit #13: 超时连接问题

Michael-Pi, 2017-01-16:
我设置了连接次数4，超时50秒，为什么还是每5,6秒就重新连接？

dingjikerbo, 2017-01-16:
连接失败了吧

Michael-Pi, 2017-01-16:
是的。无论设置超时是多少秒，都是5,6秒后retry

dingjikerbo, 2017-01-16:
你看看日志是不是失败了，失败原因是什么

Michael-Pi, 2017-01-16:
01-16 00:17:52.623 5016-5016/com.ekuaitu V/miio-bluetooth: connect 98:5D:AD:1F:04:83
01-16 00:17:52.653 5016-6286/com.ekuaitu W/miio-bluetooth: Process BleConnectRequest, status = Disconnected
01-16 00:17:52.653 5016-6286/com.ekuaitu V/miio-bluetooth: openGatt for 98:5D:AD:1F:04:83
01-16 00:17:52.653 5016-6286/com.ekuaitu D/BluetoothGatt: connect() - device: 98:5D:AD:1F:04:83, auto: false
01-16 00:17:52.653 5016-6286/com.ekuaitu D/BluetoothGatt: registerApp()
01-16 00:17:52.653 5016-6286/com.ekuaitu D/BluetoothGatt: registerApp() - UUID=eac0e634-c06b-4765-a702-b34be24df8dd
01-16 00:17:52.693 5016-5028/com.ekuaitu D/BluetoothGatt: onClientRegistered() - status=0 clientIf=7
01-16 00:17:57.703 5016-6286/com.ekuaitu V/miio-bluetooth: BleConnectRequest 98:5D:AD:1F:04:83 >>> connect timeout
01-16 00:17:57.703 5016-6286/com.ekuaitu V/miio-bluetooth: BleConnectRequest 98:5D:AD:1F:04:83 >>> close gatt
01-16 00:17:57.703 5016-6286/com.ekuaitu V/miio-bluetooth: closeGatt for 98:5D:AD:1F:04:83
01-16 00:17:57.703 5016-6286/com.ekuaitu D/BluetoothGatt: close()
01-16 00:17:57.703 5016-6286/com.ekuaitu D/BluetoothGatt: unregisterApp() - mClientIf=7
01-16 00:17:57.703 5016-6286/com.ekuaitu V/miio-bluetooth: BleConnectRequest 98:5D:AD:1F:04:83 >>> retry connect later

dingjikerbo, 2017-01-22:
超时时间你确定是50s么，应该是setTimeout(50000)

kanshenmekan, 2017-03-21:
连log都一样，老哥，找到问题在哪没 refreshCache好像没用

kanshenmekan, 2017-03-21:
我那显示是什么133  
onConnectionStateChange for 98:D3:34:90:86:F9: status = 133, newState = 0
closeGatt for 98:D3:34:90:86:F9

dingjikerbo, 2017-04-18:
133一般是设备无响应

closed 2017-04-18 by dingjikerbo