dingjikerbo/Android-BluetoothKit #25: BLE分包写

dingjikerbo, 2017-05-02:
BLE单次写的数据量大小是有限制的，通常是20字节，可以尝试通过requestMTU增大，但不保证能成功。分包写是一种解决方案，需要定义分包协议，假设每个包大小20字节，分两种包，数据包和非数据包。对于数据包，头两个字节表示包的序号，剩下的都填充数据。对于非数据包，主要是发送一些控制信息。

总体流程如下：
1、发端发送控制包，告诉对方要发送的总的包数
2、收端回一个ACK包，表示准备好了，对方可以发送了
3、发端开始批量发送数据，为了速度，不需要对端回复
4、收端收到最后一个包或超时后，开始检查还缺哪些包，然后回复ACK，带上丢的包序号
5、发端重发丢的包
6、收端全部收到，回复ACK，CRC校验，包数据拼接好了传到上层

这里面核心是状态机，另外要注意对于BLE来说，主端和从端都可以收发数据，但是是非对称的。对主端来说，发是write characteristic，收是onCharacteristicChanged，对从端来说，发是sendNotify，收是onCharacteristicWrite。尽管如此，但是协议部分是相同的，所以我们要抽象成两层，底层是收发的物理层，上层是分包的协议层，再上层就是业务层。

具体代码可参考[Channel.java](https://github.com/dingjikerbo/BluetoothKit/blob/master/library/src/main/java/com/inuker/bluetooth/library/channel/Channel.java)

jpthson, 2017-05-09:
Hi , 
I am using your bluetooth kit library , i am trying to fetch heart beat rate from ble device  using read descriptor method in your library , but its returning a value of byte of 0000 . Can you please guide me on this ??? How to fetch heart beat rate from ble device using your bluetooth kit library . Thanks in advance.

vancentvan, 2019-09-11:
您好！，如果我对端的底层协议是封闭的话，我需要怎么发送超过20字节的数据呢？谢谢