dingjikerbo/Android-BluetoothKit #28: registerBluetoothStateListener  这个监听回调不及时，有时候不回调。

yswheye, 2017-05-11:
registerBluetoothStateListener  这个监听回调不及时，有时候不回调。
但是，通过log看到BluetoothKit内部是有状态变化的

yswheye, 2017-05-11:
状态监听有点乱
这个收到及时
BluetoothReceiver.getInstance().register(new BleConnectStatusChangeListener() {
            @Override
            protected void onConnectStatusChanged(String mac, int status) {
                if (Constants.STATUS_DISCONNECTED == status) {
                   
                } else if (Constants.STATUS_CONNECTED == status) {
                  
                }
            }
        });

ClientManager.getClient().registerConnectStatusListener 收到不及时或收不到

yswheye, 2017-05-11:
但是BluetoothReceiver.getInstance().register会回调四次

dingjikerbo, 2017-05-26:
你可能注册了多次。

dingjikerbo, 2017-05-26:
你确定一下到底是registerBluetoothStateListener不及时还是registerConnectStatusListener不及时

closed 2017-07-21 by dingjikerbo