dingjikerbo/Android-BluetoothKit #30: 连接成功后，大概5 、6 秒后就自动断开了，请问是怎么回事?

androidxu, 2017-05-16:
/miio-bluetooth: setConnectStatus status = Service Ready
/miio-bluetooth: refreshServiceProfile for A0:E6:F8:51:7B:46
/miio-bluetooth: Service: 00001800-0000-1000-8000-00805f9b34fb
/miio-bluetooth: character: uuid = 00002a00-0000-1000-8000-00805f9b34fb
/miio-bluetooth: character: uuid = 00002a01-0000-1000-8000-00805f9b34fb
/miio-bluetooth: character: uuid = 00002a04-0000-1000-8000-00805f9b34fb
/miio-bluetooth: Service: 00001801-0000-1000-8000-00805f9b34fb
/miio-bluetooth: Service: 0000180a-0000-1000-8000-00805f9b34fb
/miio-bluetooth: character: uuid = 00002a25-0000-1000-8000-00805f9b34fb
/miio-bluetooth: character: uuid = 00002a26-0000-1000-8000-00805f9b34fb
/miio-bluetooth: character: uuid = 00002a27-0000-1000-8000-00805f9b34fb
/miio-bluetooth: character: uuid = 00002a28-0000-1000-8000-00805f9b34fb
/miio-bluetooth: character: uuid = 00002a29-0000-1000-8000-00805f9b34fb
/miio-bluetooth: Service: 0000fff0-0000-1000-8000-00805f9b34fb
/miio-bluetooth: character: uuid = 0000fff1-0000-1000-8000-00805f9b34fb
/miio-bluetooth: character: uuid = 0000fff2-0000-1000-8000-00805f9b34fb
/miio-bluetooth: BluetoothReceiver onReceive: action.connect_status_changed
/miio-bluetooth: onConnectStatusChanged for A0:E6:F8:51:7B:46, status = 16
/miio-bluetooth: character: uuid = 0000fff8-0000-1000-8000-00805f9b34fb
/miio-bluetooth: BleConnectRequest A0:E6:F8:51:7B:46 >>> request complete: code = 0
/BluetoothGatt: onClientConnectionState() - status=8 clientIf=6 device=A0:E6:F8:51:7B:46
/miio-bluetooth: onConnectionStateChange for A0:E6:F8:51:7B:46: status = 8, newState = 0
/miio-bluetooth: closeGatt for A0:E6:F8:51:7B:46
/BluetoothGatt: close()
/BluetoothGatt: unregisterApp() - mClientIf=6
/miio-bluetooth: setConnectStatus status = Disconnected
/miio-bluetooth: BluetoothReceiver onReceive: action.connect_status_changed
/miio-bluetooth: onConnectStatusChanged for A0:E6:F8:51:7B:46, status = 32

dingjikerbo, 2017-05-26:
手机和设备之间底层参数没协商妥当，建议设备端看看

closed 2017-05-26 by dingjikerbo

Allyns, 2020-03-19:
我也遇到了同样的问题，如果发送命令五秒内没有返回通知会自动断开！请问在哪里可以设置改变自动断开的时间？