dingjikerbo/Android-BluetoothKit #39: Notify Not getting called sometime

Mindbowser, 2017-06-21:
We are trying to use following code,

```
BleClientManager.getInstance().notify(address.trim(),
                OTA_GATT_SERVICE_UUID,
                OTA_GATT_CHARACTERIC_IDENTIFY_UUID,
                new BleNotifyResponse() {
                    @Override
                    public void onNotify(UUID service, UUID character, byte[] value) {
                        Log.i("DownloadFirmware","onNotify gets called..");
                        response.onNotify(service, character, value);
                    }

                    @Override
                    public void onResponse(int code) {
                        Log.i("BleOperations", String.format("fwUploadIdentifyCharactericNotify  imgType = onResponse %d", code));
                        Log.i("DownloadFirmware","onResponse gets called.."+code);
                        if(code == REQUEST_SUCCESS) {
                            response.firmwareUpgradeSuccess(UPLOAD_IDENTIFY_CHARACTERIC,address);
                        }
                    }
                });
```

But sometime onNotify is not getting called, I thought to use read method but that is also returning -1, any idea why it is happening sometime only? FYI, it is working flawlessly in iOS app.

Amitshinde19, 2017-06-22:
I have facing the same issue. I have getting following exception.

06-22 11:20:11.276: E/BtOppRfcommListener(4398): Error accept connection java.io.IOException: read failed, socket might closed or timeout, read ret: -1
06-22 11:20:11.777: E/BtOppRfcommListener(4398): Error accept connection java.io.IOException: read failed, socket might closed or timeout, read ret: -1
06-22 11:20:12.277: E/BtOppRfcommListener(4398): Error accept connection java.io.IOException: read failed, socket might closed or timeout, read ret: -1

if this might help
------------------------
https://stackoverflow.com/questions/18657427/ioexception-read-failed-socket-might-closed-bluetooth-on-android-4-3
https://stackoverflow.com/questions/29143695/java-io-ioexception-read-failed-socket-might-closed-or-timeout-read-ret-1-o

jpthson, 2017-07-13:
Hi  @Mindbowser,

I am using the same Notify method, Which you have mentioned in your post. I am facing a issue while notifying the Custom UUID's , I am getting a success code as response but its not returning any byte[] value. But the same code is working fine when I used to fetch the Heart Rate values.  Do you have any idea on this ? Thanks in Advance!.

jpthson, 2017-07-17:
Hi ,

If anyone is facing a issue on notifying the custom service, please find the below solution code.

copy and paste the below method in BleConnectWorker.class

 public void exchangeGattMtu(int mtu) {
          if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.LOLLIPOP) {
              int retry = 6;
              boolean status = false;
              while (!status && retry > 0) {

                  status = mBluetoothGatt.requestMtu(mtu);

                  retry--;

              }

          }

    }

And call the below line inside the  broadcastCharacterChanged method in the same class (BleConnectWorker.class).

exchangeGattMtu(104);

Amitshinde19, 2017-08-28:
public void exchangeGattMtu(int mtu) {
if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.LOLLIPOP) {
int retry = 6;
boolean status = false;
while (!status && retry > 0) {

              status = mBluetoothGatt.requestMtu(mtu);

              retry--;

          }

      }

}

exchangeGattMtu(104);

Changed above code in BleConnectWorker class but still it is not calling onNotify().