dingjikerbo/Android-BluetoothKit #44: 红米note4x android 6.0.1 MIUI8.5稳定版 每次连接都会被断掉，重试了好几次才有可能连上

mengcd, 2017-07-17:
12:50:47.404 25183-25183/com.inuker.bluetooth I/Timeline: Timeline: Activity_launch_request time:3845900
12:50:47.425 25183-25183/com.inuker.bluetooth V/miio-bluetooth: stopSearch
12:50:47.452 25183-25183/com.inuker.bluetooth V/miio-bluetooth: connect 08:7C:BE:07:D2:23
12:50:47.456 25183-25183/com.inuker.bluetooth D/ActivityThreadInjector: clearCachedDrawables.
12:50:47.500 25183-26077/com.inuker.bluetooth W/miio-bluetooth: Process BleConnectRequest, status = Disconnected
12:50:47.502 25183-26077/com.inuker.bluetooth V/miio-bluetooth: openGatt for 08:7C:BE:07:D2:23
12:50:47.502 25183-26077/com.inuker.bluetooth D/BluetoothGatt: connect() - device: 08:7C:BE:07:D2:23, auto: false
12:50:47.502 25183-26077/com.inuker.bluetooth D/BluetoothGatt: registerApp()
12:50:47.502 25183-26077/com.inuker.bluetooth D/BluetoothGatt: registerApp() - UUID=a0b100ba-f595-43f0-906e-5f01c022fa41
12:50:47.505 25183-26056/com.inuker.bluetooth D/BluetoothGatt: onClientRegistered() - status=0 clientIf=5
12:50:48.732 25183-5740/com.inuker.bluetooth D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=5 device=08:7C:BE:07:D2:23
12:50:48.733 25183-26077/com.inuker.bluetooth V/miio-bluetooth: onConnectionStateChange for 08:7C:BE:07:D2:23: status = 133, newState = 0
12:50:48.733 25183-26077/com.inuker.bluetooth V/miio-bluetooth: closeGatt for 08:7C:BE:07:D2:23
12:50:48.733 25183-26077/com.inuker.bluetooth D/BluetoothGatt: close()
12:50:48.733 25183-26077/com.inuker.bluetooth D/BluetoothGatt: unregisterApp() - mClientIf=5
12:50:48.734 25183-26077/com.inuker.bluetooth V/miio-bluetooth: BleConnectRequest 08:7C:BE:07:D2:23 >>> retry connect later
12:50:48.734 25183-26077/com.inuker.bluetooth V/miio-bluetooth: setConnectStatus status = Disconnected
12:50:48.747 25183-25183/com.inuker.bluetooth V/miio-bluetooth: BluetoothReceiver onReceive: action.connect_status_changed
12:50:48.749 25183-25183/com.inuker.bluetooth V/miio-bluetooth: onConnectStatusChanged for 08:7C:BE:07:D2:23, status = 32
12:50:48.764 25183-25183/com.inuker.bluetooth V/miio-bluetooth: DeviceDetailActivity onConnectStatusChanged 32 in main
12:50:48.765 25183-25183/com.inuker.bluetooth V/miio-bluetooth: connect 08:7C:BE:07:D2:23
12:50:49.735 25183-26077/com.inuker.bluetooth V/miio-bluetooth: openGatt for 08:7C:BE:07:D2:23
12:50:49.736 25183-26077/com.inuker.bluetooth D/BluetoothGatt: connect() - device: 08:7C:BE:07:D2:23, auto: false
12:50:49.736 25183-26077/com.inuker.bluetooth D/BluetoothGatt: registerApp()
12:50:49.737 25183-26077/com.inuker.bluetooth D/BluetoothGatt: registerApp() - UUID=96b34614-408c-48a3-9426-c2c16cf65f16
12:50:49.744 25183-5740/com.inuker.bluetooth D/BluetoothGatt: onClientRegistered() - status=0 clientIf=5
12:50:50.002 25183-25197/com.inuker.bluetooth D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=5 device=08:7C:BE:07:D2:23
12:50:50.002 25183-26077/com.inuker.bluetooth V/miio-bluetooth: onConnectionStateChange for 08:7C:BE:07:D2:23: status = 133, newState = 0
12:50:50.002 25183-26077/com.inuker.bluetooth V/miio-bluetooth: closeGatt for 08:7C:BE:07:D2:23
12:50:50.002 25183-26077/com.inuker.bluetooth D/BluetoothGatt: close()
12:50:50.003 25183-26077/com.inuker.bluetooth D/BluetoothGatt: unregisterApp() - mClientIf=5
12:50:50.005 25183-26077/com.inuker.bluetooth V/miio-bluetooth: BleConnectRequest 08:7C:BE:07:D2:23 >>> retry connect later
12:50:50.005 25183-26077/com.inuker.bluetooth V/miio-bluetooth: setConnectStatus status = Disconnected
12:50:50.017 25183-25183/com.inuker.bluetooth V/miio-bluetooth: BluetoothReceiver onReceive: action.connect_status_changed
12:50:50.017 25183-25183/com.inuker.bluetooth V/miio-bluetooth: onConnectStatusChanged for 08:7C:BE:07:D2:23, status = 32
12:50:50.030 25183-25183/com.inuker.bluetooth V/miio-bluetooth: DeviceDetailActivity onConnectStatusChanged 32 in main
12:50:50.031 25183-25183/com.inuker.bluetooth V/miio-bluetooth: connect 08:7C:BE:07:D2:23
12:50:51.006 25183-26077/com.inuker.bluetooth V/miio-bluetooth: openGatt for 08:7C:BE:07:D2:23
12:50:51.007 25183-26077/com.inuker.bluetooth D/BluetoothGatt: connect() - device: 08:7C:BE:07:D2:23, auto: false
12:50:51.007 25183-26077/com.inuker.bluetooth D/BluetoothGatt: registerApp()
12:50:51.007 25183-26077/com.inuker.bluetooth D/BluetoothGatt: registerApp() - UUID=5bd5dbc5-b8c2-4fef-9697-734412d77fde
12:50:51.011 25183-25197/com.inuker.bluetooth D/BluetoothGatt: onClientRegistered() - status=0 clientIf=5
12:50:51.267 25183-5740/com.inuker.bluetooth D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=5 device=08:7C:BE:07:D2:23
12:50:51.271 25183-26077/com.inuker.bluetooth V/miio-bluetooth: onConnectionStateChange for 08:7C:BE:07:D2:23: status = 133, newState = 0
12:50:51.271 25183-26077/com.inuker.bluetooth V/miio-bluetooth: closeGatt for 08:7C:BE:07:D2:23
12:50:51.271 25183-26077/com.inuker.bluetooth D/BluetoothGatt: close()
12:50:51.271 25183-26077/com.inuker.bluetooth D/BluetoothGatt: unregisterApp() - mClientIf=5
12:50:51.272 25183-26077/com.inuker.bluetooth V/miio-bluetooth: BleConnectRequest 08:7C:BE:07:D2:23 >>> retry connect later
12:50:51.272 25183-26077/com.inuker.bluetooth V/miio-bluetooth: setConnectStatus status = Disconnected
12:50:51.282 25183-25183/com.inuker.bluetooth V/miio-bluetooth: BluetoothReceiver onReceive: action.connect_status_changed
12:50:51.282 25183-25183/com.inuker.bluetooth V/miio-bluetooth: onConnectStatusChanged for 08:7C:BE:07:D2:23, status = 32
12:50:51.297 25183-25183/com.inuker.bluetooth V/miio-bluetooth: DeviceDetailActivity onConnectStatusChanged 32 in main
12:50:51.298 25183-25183/com.inuker.bluetooth V/miio-bluetooth: connect 08:7C:BE:07:D2:23
12:50:52.274 25183-26077/com.inuker.bluetooth V/miio-bluetooth: openGatt for 08:7C:BE:07:D2:23
12:50:52.275 25183-26077/com.inuker.bluetooth D/BluetoothGatt: connect() - device: 08:7C:BE:07:D2:23, auto: false
12:50:52.275 25183-26077/com.inuker.bluetooth D/BluetoothGatt: registerApp()
12:50:52.275 25183-26077/com.inuker.bluetooth D/BluetoothGatt: registerApp() - UUID=2c22c7a6-22eb-4c1f-9bda-be8203bdad18
12:50:52.286 25183-25199/com.inuker.bluetooth D/BluetoothGatt: onClientRegistered() - status=0 clientIf=5
12:50:52.536 25183-5740/com.inuker.bluetooth D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=5 device=08:7C:BE:07:D2:23
12:50:52.544 25183-26077/com.inuker.bluetooth V/miio-bluetooth: onConnectionStateChange for 08:7C:BE:07:D2:23: status = 133, newState = 0
12:50:52.544 25183-26077/com.inuker.bluetooth V/miio-bluetooth: closeGatt for 08:7C:BE:07:D2:23
12:50:52.545 25183-26077/com.inuker.bluetooth D/BluetoothGatt: close()
12:50:52.545 25183-26077/com.inuker.bluetooth D/BluetoothGatt: unregisterApp() - mClientIf=5
12:50:52.547 25183-26077/com.inuker.bluetooth V/miio-bluetooth: BleConnectRequest 08:7C:BE:07:D2:23 >>> request complete: code = -1
12:50:52.548 25183-26077/com.inuker.bluetooth V/miio-bluetooth: setConnectStatus status = Disconnected
12:50:52.558 25183-26077/com.inuker.bluetooth W/miio-bluetooth: Process BleConnectRequest, status = Disconnected
12:50:52.560 25183-26077/com.inuker.bluetooth V/miio-bluetooth: openGatt for 08:7C:BE:07:D2:23
12:50:52.561 25183-26077/com.inuker.bluetooth D/BluetoothGatt: connect() - device: 08:7C:BE:07:D2:23, auto: false
12:50:52.561 25183-26077/com.inuker.bluetooth D/BluetoothGatt: registerApp()
12:50:52.561 25183-26077/com.inuker.bluetooth D/BluetoothGatt: registerApp() - UUID=8624220e-5442-4e03-a280-1f5a2c63439c
[... 61 lines cut ...]
12:50:56.058 25183-26077/com.inuker.bluetooth V/miio-bluetooth: character: uuid = d44bc439-abfd-45a2-b575-925416129600
12:50:56.058 25183-26077/com.inuker.bluetooth V/miio-bluetooth: character: uuid = d44bc439-abfd-45a2-b575-925416129601
12:50:56.058 25183-26077/com.inuker.bluetooth V/miio-bluetooth: BleConnectRequest 08:7C:BE:07:D2:23 >>> request complete: code = 0
12:50:56.064 25183-25183/com.inuker.bluetooth V/miio-bluetooth: BluetoothReceiver onReceive: action.connect_status_changed
12:50:56.064 25183-25183/com.inuker.bluetooth V/miio-bluetooth: onConnectStatusChanged for 08:7C:BE:07:D2:23, status = 16
12:50:56.068 25183-26077/com.inuker.bluetooth W/miio-bluetooth: Process BleConnectRequest, status = Service Ready
12:50:56.070 25183-26077/com.inuker.bluetooth V/miio-bluetooth: BleConnectRequest 08:7C:BE:07:D2:23 >>> request complete: code = 0
12:50:56.080 25183-26077/com.inuker.bluetooth W/miio-bluetooth: Process BleConnectRequest, status = Service Ready
[... cut ...]
                                                                >>> Character: BleGattCharacter{uuid=d44bc439-abfd-45a2-b575-925416129600, property=12, permissions=0, descriptors=[BleGattDescriptor{mUuid=00002901-0000-1000-8000-00805f9b34fb, mPermissions=0, mValue=null}]}
                                                                >>> Character: BleGattCharacter{uuid=d44bc439-abfd-45a2-b575-925416129601, property=16, permissions=0, descriptors=[BleGattDescriptor{mUuid=00002902-0000-1000-8000-00805f9b34fb, mPermissions=0, mValue=null}]}

LuckyStar521, 2017-07-31:
这是硬件那边代码的问题，我们也遇到过同样的问题

Sabirjan, 2017-08-07:
我也遇到同样的问题
连接成功后，调用
 mBluetoothGatt = device.connectGatt( m_ctx, false, mGattCallback);
这个方法就无法连接...

mengcd, 2017-08-17:
@LuckyStar521 后来有解决了吗，有的话，是如何解决的呢；）

LuckyStar521, 2017-08-18:
解决了，是硬件设备请求了Mtu导致，红米note4X不支持连接的设备请求Mtu，一请求就会断开，硬件把请求Mtu的代码去掉就好了，若要是改变Mtu大小，手机主动requestMtu即可

Sabirjan, 2017-08-18:
@LuckyStar521 不太明白。能不能贴一下代码，谢谢

LuckyStar521, 2017-08-19:
代码是硬件改的，你找他们改的就行了