dingjikerbo/Android-BluetoothKit #74: 多进程，主进程搜索过程中，其他进程突然崩溃（偶发）

niulinguo, 2017-11-22:
E/AndroidRuntime: FATAL EXCEPTION: main
                                                                          Process: com.huabao.ttsdk.app:tic, PID: 12222
                                                                          java.lang.NullPointerException: Attempt to read from field 'long android.os.Parcel.mNativePtr' on a null object reference
                                                                              at android.os.Parcel.readException(Parcel.java:1626)
                                                                              at android.os.Parcel.readException(Parcel.java:1573)
                                                                              at com.inuker.bluetooth.library.IResponse$Stub$Proxy.onResponse(IResponse.java:103)
                                                                              at com.inuker.bluetooth.library.BluetoothServiceImpl$1.onResponse(BluetoothServiceImpl.java:76)
                                                                              at com.inuker.bluetooth.library.BluetoothServiceImpl$1.onResponse(BluetoothServiceImpl.java:67)
                                                                              at com.inuker.bluetooth.library.search.BluetoothSearchManager$1.onDeviceFounded(BluetoothSearchManager.java:31)
                                                                              at com.inuker.bluetooth.library.search.BluetoothSearchHelper$BluetoothSearchResponseImpl.onDeviceFounded(BluetoothSearchHelper.java:74)
                                                                              at com.inuker.bluetooth.library.search.BluetoothSearchRequest.handleMessage(BluetoothSearchRequest.java:63)
                                                                              at android.os.Handler.dispatchMessage(Handler.java:98)
                                                                              at android.os.Looper.loop(Looper.java:158)
                                                                              at android.app.ActivityThread.main(ActivityThread.java:7237)
                                                                              at java.lang.reflect.Method.invoke(Native Method)
                                                                              at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1230)
                                                                              at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1120)

niulinguo, 2017-11-23:
修正一下问题描述

将BluetoothService注册到其他子进程，在主进程搜索设备的时候，会有崩溃（20%偶发率）

```
<service
            android:name="com.inuker.bluetooth.library.BluetoothService"
            android:process=":tic"
            />
```

BestCoderXQX, 2018-01-25:
请问你这问题解决了么  我也出现这问题了 小部分机型概率出现

niulinguo, 2018-01-25:
BluetoothService服务注册到哪个进程，哪个进程才能用这个服务，不会出现崩溃。

BestCoderXQX, 2018-01-26:
可以加下qq 交流下么  1057000259