dingjikerbo/Android-BluetoothKit #75: 问题一：android 6.0及以上，连接ble设备，设备不显示在系统设置中的蓝牙中，但是android5.0就有

HanHuoBin, 2017-11-22:


HanHuoBin, 2017-11-22:
问题一：android 6.0及以上，连接ble设备，设备不显示在系统设置中的蓝牙中，但是android5.0就有 
问题二：在基于第一个问题上，我们的硬件工程师把安全配对这块去掉，发送请求硬件可以收到，但是多次请求之后会出现REQUEST_OVERFLOW，不知道这是个什么问题，一旦清空request又出现REQUEST_CANCELED

HanHuoBin, 2017-11-22:
问题二已经找到解决方案了，对于请求来说，完整的一次请求之后去clear才是正确的操作
遇到的问题1：当你分包发送的时候，需要在相邻的两个包之间加点延时，不然会出现一个包还没到，另一个就开发送了
遇到的问题2：对于遇到REQUEST_OVERFLOW的问题，我们判断当前的request是不是某个请求的最后一个包，如果是最后一个包，在你发送成功之后清空REQUEST_WRITE，如果其中某个包未发送成功，继续调用write方法
```
private void write(int j,UUID uuid, UUID characUuid, byte[] content) {
        mClient.write(address, uuid, characUuid, content,
                new BleWriteResponse() {
                    @Override
                    public void onResponse(int code) {
                        Log.i("BlueDevice", "code:" + code);
                        if (code == Constants.REQUEST_SUCCESS) {
                            if(j == 2){
                                mClient.clearRequest(model.getAddress(), Constants.REQUEST_WRITE);
                                T.showShort("发送成功");
                            }
                        } else if (code == Constants.REQUEST_OVERFLOW) {
                            if(j == 2){
                                mClient.clearRequest(model.getAddress(), Constants.REQUEST_WRITE);
                            }else{
                                write(j,uuid, characUuid, content);
                            }
                        }
                    }

                });
        try {
            Thread.sleep(80);
        } catch (InterruptedException e) {

        }
    }
```
说明：因为我这边是分两个包发送的，所以用2来做是否结束的判断

zetaoWu, 2018-01-05:
你好，看到您的问题 想请教一下，你们连接的是ble 会出现配对请求吗？ 我看网上说 ble 不会出现配对弹窗。但是我连接的ble设备会出现配对弹窗确认。

HanHuoBin, 2018-01-05:
蓝牙低功耗没有配对弹窗的哦，你看看这个配对框是系统的蓝牙配对框还是你的代码写的

g19980115, 2018-03-16:
@zetaoWu 低功耗蓝牙也有需要配对的，只是很少，要看硬件工程师怎么做的

qwuip789, 2024-05-29:
> @zetaoWu 低功耗蓝牙也有需要配对的，只是很少，要看硬件工程师怎么做的
怎么实现请求配对