dingjikerbo/Android-BluetoothKit #119: startLeScan(): null

webloverand, 2019-03-13:
设备 : 华为P20 Pro

输出如下 :
2019-03-13 18:30:46.755 10933-10933/com.inuker.bluetooth W/miio-bluetooth: MainActivity.onSearchStarted
2019-03-13 18:30:46.865 10933-10933/com.inuker.bluetooth V/miio-bluetooth: Ble search (10s) onSearchStarted
2019-03-13 18:30:46.865 10933-10933/com.inuker.bluetooth D/BluetoothAdapter: startLeScan(): null
2019-03-13 18:30:46.865 10933-10933/com.inuker.bluetooth I/BluetoothAdapter: getBluetoothLeScanner
2019-03-13 18:30:46.869 10933-10933/com.inuker.bluetooth D/BluetoothAdapter: isLeEnabled(): ON
2019-03-13 18:30:46.875 10933-10959/com.inuker.bluetooth D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=7 mScannerId=0
2019-03-13 18:30:46.987 10933-10933/com.inuker.bluetooth D/HwAppInnerBoostImpl: asyncReportData com.inuker.bluetooth,1,1,2,0 interval=591
2019-03-13 18:30:48.729 10933-10933/com.inuker.bluetooth D/HwAppInnerBoostImpl: asyncReportData com.inuker.bluetooth,1,1,3,0 interval=2333
2019-03-13 18:30:48.984 10933-10933/com.inuker.bluetooth D/HwAppInnerBoostImpl: asyncReportData com.inuker.bluetooth,1,1,4,0 interval=2588
2019-03-13 18:30:56.898 10933-10933/com.inuker.bluetooth D/BluetoothAdapter: stopLeScan()
2019-03-13 18:30:56.898 10933-10933/com.inuker.bluetooth I/BluetoothAdapter: getBluetoothLeScanner
2019-03-13 18:30:56.902 10933-10933/com.inuker.bluetooth D/BluetoothAdapter: isLeEnabled(): ON
2019-03-13 18:30:56.915 10933-10933/com.inuker.bluetooth V/miio-bluetooth: Ble search (10s) onSearchStopped
2019-03-13 18:30:57.019 10933-10933/com.inuker.bluetooth V/miio-bluetooth: Ble search (10s) onSearchStarted
2019-03-13 18:30:57.019 10933-10933/com.inuker.bluetooth D/BluetoothAdapter: startLeScan(): null
2019-03-13 18:30:57.019 10933-10933/com.inuker.bluetooth I/BluetoothAdapter: getBluetoothLeScanner
2019-03-13 18:30:57.023 10933-10933/com.inuker.bluetooth D/BluetoothAdapter: isLeEnabled(): ON

nihuajun1991, 2020-08-03:
解决了吗？我在小米k30上也出现这个问题，但是其他手机没问题

huangxiongbiao12, 2023-09-12:
安卓13 遇到这个问题  有解决吗