dingjikerbo/Android-BluetoothKit #122: 这个项目不维护了吗，需要设置MTU

SmartZane, 2019-04-08:
这个项目不维护了吗，需要设置MTU

Helloneinei, 2019-05-10:
requestMtu(String mac, int mtu, final BleMtuResponse response)；
看项目的时候看仔细一点，更新记录不是有吗.

huihui2286433852, 2019-07-02:
我这也有这需求    大佬有空研究研究

huihui2286433852, 2019-07-02:
我试着去改了一下    发现有一些设备一直再重连    就是断一下连一下

Wizardp007, 2019-07-08:
> 我这也有这需求 大佬有空研究研究

`public void requestMtu(String mac, int mtu, BleMtuResponse response)`

这个功能真的有。。。另外，我发现在8.0以上订阅了各种连接监听之后，去断开会特别慢，博主让我横纵向对比，我现在还没来得及测呢，你们有没有发现类似的问题？可以一起交流一下

weizhizhu, 2020-04-27:
> 我试着去改了一下 发现有一些设备一直再重连 就是断一下连一下

这个问题有解决的方案吗？？

Wizardp007, 2020-04-27:
> > 我试着去改了一下 发现有一些设备一直再重连 就是断一下连一下
> 
> 这个问题有解决的方案吗？？

这问题还是跟硬件对一下吧，看一下你们设备支持的mtu是多少，然后可以换个测试机试一下，有些机型的mtu也不能太大

weizhizhu, 2020-04-27:
> > > 我试着去改了一下 发现有一些设备一直再重连 就是断一下连一下
> > 
> > 
> > 这个问题有解决的方案吗？？
> 
> 这问题还是跟硬件对一下吧，看一下你们设备支持的mtu是多少，然后可以换个测试机试一下，有些机型的mtu也不能太大

但是我并没有去设置MTU的大小，发现蓝牙设备也是不断断开重连，有遇到过吗

Wizardp007, 2020-04-27:
> > > > 我试着去改了一下 发现有一些设备一直再重连 就是断一下连一下
> > > 
> > > 
> > > 这个问题有解决的方案吗？？
> > 
> > 
> > 这问题还是跟硬件对一下吧，看一下你们设备支持的mtu是多少，然后可以换个测试机试一下，有些机型的mtu也不能太大
> 
> 但是我并没有去设置MTU的大小，发现蓝牙设备也是不断断开重连，有遇到过吗

有，找到问题是因为我们硬件的事儿，框架本身没有问题，从使用逻辑或硬件开始对吧

weizhizhu, 2020-04-28:
> > > > > 我试着去改了一下 发现有一些设备一直再重连 就是断一下连一下
> > > > 
> > > > 
> > > > 这个问题有解决的方案吗？？
> > > 
> > > 
> > > 这问题还是跟硬件对一下吧，看一下你们设备支持的mtu是多少，然后可以换个测试机试一下，有些机型的mtu也不能太大
> > 
> > 
> > 但是我并没有去设置MTU的大小，发现蓝牙设备也是不断断开重连，有遇到过吗
> 
> 有，找到问题是因为我们硬件的事儿，框架本身没有问题，从使用逻辑或硬件开始对吧

好的 感谢感谢

shualai, 2021-06-15:
> 这个项目不维护了吗，需要设置MTU

大佬你解决这个问题了嘛?我也需要设置MTU