dingjikerbo/Android-BluetoothKit #124: 已连接设备问题？！

SilverTiTan, 2019-06-25:
蓝牙原生是有device.connectGatt()方法返回BluetoothGatt对象这个BluetoothGatt对象可以调用到bluetoothGatt.getDevice().getName()已连接设备 而我目前在框架里还没有找到这个方法 希望作者可以加一下 辛苦了

Wizardp007, 2019-07-11:
> 蓝牙原生是有device.connectGatt()方法返回BluetoothGatt对象这个BluetoothGatt对象可以调用到bluetoothGatt.getDevice().getName()已连接设备 而我目前在框架里还没有找到这个方法 希望作者可以加一下 辛苦了

BluetoothSearchRequest 这个类里有以下方法：
`notifyConnectedBluetoothDevices()`
` notifyConnectedBluetoothLeDevices()`
`notifyBondedBluetoothClassicDevices()`
你可以在源码上修改，去实现你的需求，但是建议你前后结构一定要明白，尽量不要改源码，也可以
直接在蓝牙管理类里BluetoothUtils： 
`getConnectedBluetoothLeDevices()` 
会返回手机已连接的BLE设备，这个能满足你的需求吧？

SilverTiTan, 2019-07-18:
你说的getConnectedBluetoothLeDevices()方法 返回的是BluetoothDevice类 而框架需要的是SearchResult。当连接退出页面 在进入连接页面没有操作并退出后 会造成IllegalArgumentException:  is not a valid Bluetooth address。

closed 2019-07-18 by SilverTiTan