dingjikerbo/Android-BluetoothKit #125: android.os.DeadSystemException  com.inuker.bluetooth.library.utils.BluetoothUtils.isBleSupported(BluetoothUtils.java:80)

852172891, 2019-06-28:
log
```

java.lang.RuntimeException:android.os.DeadSystemException
--
2 android.app.ApplicationPackageManager.hasSystemFeature(ApplicationPackageManager.java:501)
3 ......
4 Caused by:
5 android.os.DeadSystemException:
6 android.app.ApplicationPackageManager.hasSystemFeature(ApplicationPackageManager.java:501)
7 android.app.ApplicationPackageManager.hasSystemFeature(ApplicationPackageManager.java:493)
8 com.inuker.bluetooth.library.utils.BluetoothUtils.isBleSupported(BluetoothUtils.java:80)
9 com.inuker.bluetooth.library.connect.request.BleRequest.process(BleRequest.java:156)
10 com.inuker.bluetooth.library.connect.BleConnectDispatcher.scheduleNextRequest(BleConnectDispatcher.java:203)
11 com.inuker.bluetooth.library.connect.BleConnectDispatcher.handleMessage(BleConnectDispatcher.java:218)
12 android.os.Handler.dispatchMessage(Handler.java:98)

```

Wizardp007, 2019-07-11:
> log
>
> ```
>
> java.lang.RuntimeException:android.os.DeadSystemException
> --
> 2 android.app.ApplicationPackageManager.hasSystemFeature(ApplicationPackageManager.java:501)
> 3 ......
> 4 Caused by:
> 5 [...]

蓝牙相关权限都申请了吗？

852172891, 2019-07-11:
都申请了
[image]

Lebron-Liu, 2024-10-21:
解决了吗？是不是Android 14兼容问题