dingjikerbo/Android-BluetoothKit #130: 使用连接方法，只能配对成功，不能连接成功

mister-tang, 2019-10-21:
你好，我使用下面的连接方法，但是一直是配对成功，并没有连接成功，应该怎么处理才能跟系统蓝牙一样

BleConnectOptions options = new BleConnectOptions.Builder()
        .setConnectRetry(3)   // 连接如果失败重试3次
        .setConnectTimeout(30000)   // 连接超时30s
        .setServiceDiscoverRetry(3)  // 发现服务如果失败重试3次
        .setServiceDiscoverTimeout(20000)  // 发现服务超时20s
        .build();

mClient.connect(MAC, options, new BleConnectResponse() {
    @Override
    public void onResponse(int code, BleGattProfile data) {

    }
});

Wizardp007, 2019-10-22:
你在操作什么设备，BLE还是经典

mister-tang, 2019-10-22:
> 你在操作什么设备，BLE还是经典

android7.1+蓝牙鼠标（市面上普通的），用ble或者经典都能扫描到，并且能配对，但设备提示灯一直是闪烁，如果用系统蓝牙连接，就能连接成功

mensaoselang, 2019-10-31:
Ble的蓝牙键盘一直连接不成功，也是系统蓝牙连接成功的。connect返回code -1

Wizardp007, 2019-11-07:
问题解决了吗？我看上面这位兄弟说蓝牙键盘用HID协议，你这个蓝牙鼠标应该跑不了吧？

mister-tang, 2019-11-08:
> 问题解决了吗？我看上面这位兄弟说蓝牙键盘用HID协议，你这个蓝牙鼠标应该跑不了吧？

没有解决，问题在系统版本7.1才会出现这个bug，其他系统版本没有问题，正在研究7.1的系统蓝牙