dingjikerbo/Android-BluetoothKit #147: onNotify不回调

Kbaseczy, 2020-03-24:
AppApplication.mBluetoothClient.notify(mMac, serviceUUID, characterUUID, new BleNotifyResponse() {
            @Override
            public void onNotify(UUID service, UUID character, byte[] value) {
                LogUtil.v("data:" + ByteUtils.byteToString(value));
            }

            @Override
            public void onResponse(int code) {
                if (code == REQUEST_SUCCESS)
                    LogUtil.v("code:" + code);
            }
        });
  

如图，在回调方法onNotify没有回调，无数据。在连接成功后直接设置通知的。

penglei60, 2020-05-25:
请问解决这个问题了吗？  我也遇到了，华为手机第一次调用onNotify，code 返回0，有数据。断连，再次调用 code=-1，没有返回数据。在小米手机上，第一次调用onNdtify，code 返回0，也没有返回数据。

Kbaseczy, 2020-05-25:
我之前没回调数据是因为我把蓝牙协议写错了。你可以先用蓝牙测试工具调测一下看有没得数据返回。如果工具测试还是没得数据返回就确认下自己协议写错没，还是不行就得问问蓝牙开发了。
调测工具：MineW 串口助手--商店直接就能搜到，或者有个nrf connect。

penglei60, 2020-05-25:
nrf connect测试过，是可以连接通的，我测试小米手机是红米note4，网上讲是MTU的原因，还在研究中。但是，如果第一次调用onNotify，code 返回0，再断连，重新连接蓝牙，code 必返回-1或者-2，这个是什么原因？

Cloudant-android, 2020-11-03:
在重连之后把上一个连接的notify关闭之后再次连接，出现code=0，但是部分通知接受不到，这个不知是什么原因，我重连的代码和连接的代码使用的是一套，每次打开notify的之前都调用了一次关闭notify

Cloudant-android, 2020-11-03:
> nrf connect测试过，是可以连接通的，我测试小米手机是红米note4，网上讲是MTU的原因，还在研究中。但是，如果第一次调用onNotify，code 返回0，再断连，重新连接蓝牙，code 必返回-1或者-2，这个是什么原因？

在使用notify的时候需要关闭之前的notify，这样就不会出现发送一条消息接受多条一样的消息，也不会出现code不等于1的情况吧，我目前测试的没有出现过