dingjikerbo/Android-BluetoothKit #174: 库很久没有更新了。不维护了嘛

liqiangCoding, 2022-06-10:


fuhaodev, 2022-06-10:
用这个库吧，非常强大，支持分包、等待通知、各种需求都能满足，源码写的也是相当优雅
https://github.com/NordicSemiconductor/Android-BLE-Library

liqiangCoding, 2022-06-10:
> 用这个库吧，非常强大，支持分包、等待通知、各种需求都能满足，巧妙解决了Android BLE的各种问题，源码写的也是相当优雅 https://github.com/NordicSemiconductor/Android-BLE-Library

可以的。大佬。马上用上

fuhaodev, 2022-06-10:
嗯，非常推荐，我看了几乎所有的BLE库，只有这个源码写的是最好的，而且解决了BLE的所有问题，另外还支持自动重连

darkengine, 2023-05-06:
> 

这个库有解决Android 12以上的手机扫描BLE设备无法获取设备名字的问题吗？

darkengine, 2023-05-06:
> 嗯，非常推荐，我看了几乎所有的BLE库，只有这个https://github.com/NordicSemiconductor/Android-BLE-Library 源码写的是最好的，而且解决了BLE的所有问题，另外还支持自动重连

刚才忘了 quote了 ： 这个库有解决Android 12以上的手机扫描BLE设备无法获取设备名字的问题吗？

wudouxingjun, 2023-05-17:
> [](https://github.com/NordicSemiconductor/Android-BLE-Library)

有使用方法介绍文档吗,好像资料很少

amelons, 2023-06-27:
> 
你大爷，连附近蓝牙都搜不到，折腾半个小时跑起来，牛吹死算了