dingjikerbo/Android-BluetoothKit #186: 我已经适配好了Android 12~14

ZouZLong, 2023-11-03:


ZouZLong, 2023-11-03:
private static void registerGlobalReceiver(BroadcastReceiver receiver, IntentFilter filter) {
        try {
            LogUtil.loge("666", "Build.VERSION.SDK_INT:" + Build.VERSION.SDK_INT);//34
            LogUtil.loge("666", "Build.VERSION_CODES.O:" + Build.VERSION_CODES.O);//26
            LogUtil.loge("666", "Build.VERSION_CODES.TIRAMISU:" + Build.VERSION_CODES.TIRAMISU);//33

            if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.O) {
                if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.TIRAMISU) {
                    getContext().registerReceiver(receiver, filter, RECEIVER_EXPORTED);
                } else getContext().registerReceiver(receiver, filter);
            } else getContext().registerReceiver(receiver, filter);
        } catch (Exception e) {
            LogUtil.loge("registerGlobalReceiver:" + e);

            if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.TIRAMISU)
                getContext().registerReceiver(receiver, filter, RECEIVER_EXPORTED);
            else getContext().registerReceiver(receiver, filter);
        }

    }

修改为这样的代码就可以了

qwuip789, 2024-05-29:
> private static void registerGlobalReceiver(BroadcastReceiver receiver, IntentFilter filter) { try { LogUtil.loge("666", "Build.VERSION.SDK_INT:" + Build.VERSION.SDK_INT);//34 LogUtil.loge("666", [...]
能够说下你怎么适配的吗？