dingjikerbo/Android-BluetoothKit #190: Could not find com.inuker.bluetooth:library:1.4.0.

Priteshapatel, 2024-08-31:
I used to use this library without issues but since July 2024, I have been getting following error while compiling in Android:
"Could not find com.inuker.bluetooth:library:1.4.0."

IliaGoncharuk, 2024-09-02:
> I used to use this library without issues but since July 2024, I have been getting following error while compiling in Android: "Could not find com.inuker.bluetooth:library:1.4.0."

same

IliaGoncharuk, 2024-09-02:
> I used to use this library without issues but since July 2024, I have been getting following error while compiling in Android: "Could not find com.inuker.bluetooth:library:1.4.0."

Download it separatly and add as a aar library

tohel17, 2024-10-10:
Hi @IliaGoncharuk, 
Where can I get the aar file.

jingyao6, 2024-11-25:
> Hi @IliaGoncharuk, Where can I get the aar file.

You can use jar files through Eclipse

Ahmed-Galal85, 2025-01-02:
Where can I download the "aar" file?